mriscoc/Special_Configurations #1: [Feature request] Support for auto leveling sensor from Substancia 3D

RPI-PMK, 2022-02-15:


mriscoc, 2022-02-16:
OK, the configurations that they provided are for an old Marlin version. I'm now generating the current configuration files and posting a release for you.

mriscoc, 2022-02-16:
Your release is ready, please test it and report.
https://github.com/mriscoc/Marlin_Configurations/releases/tag/20220216-AutoLev

mriscoc, 2022-03-04:
Closing because no more feedback. You are welcome.

RPI-PMK, 2022-03-04:
Sorry, I've been very busy with work and home so unable to test yet.

Should be able to in a few weeks. Should we keep this open or open a new
case once the result is known??

mriscoc, 2022-03-04:
re-open by request.

Tillerman6, 2022-03-05:
My printer stopping is apparently not related to your fine software. The
problem repeats with a straight marlin standard build and no CR / BL
touch.  Could be a bad power supply or heater cartridge- troubleshooting
now- only fails for long prints.  Keep up the good work!  I may be back
when I get this nailed down.

mriscoc, 2022-03-05:
There is a known bug with THERMAL_PROTECTION_VARIANCE_MONITOR that produce random false thermal alerts, I will to update your special release with that feature disabled.

mriscoc, 2022-03-05:
Your special release has been updated with that feature disabled, please re-download and flash the new version.

RPI-PMK, 2022-03-06:
Sorry, very new to this.

I tried to download to test over the weekend but could not find the bin file but lots of other files and directories. What am I doing wrong???

mriscoc, 2022-03-07:
Latest version here: https://github.com/mriscoc/Marlin_Configurations/releases/tag/AutoLev

Nbrooks639, 2022-03-07:
Im still not really understanding what to do   Sent from Mail for Windows From: Miguel Risco-CastilloSent: Monday, March 7, 2022 12:11 AMTo: mriscoc/Marlin_ConfigurationsCc: Nbrooks639; ManualSubject: Re: [mriscoc/Marlin_Configurations] Support for "AutoLev" sensor from Substancia 3D (Issue #1) Latest version here: https://github.com/mriscoc/Marlin_Configurations/releases/tag/AutoLev

mriscoc, 2022-03-07:
Download the appropriate binary file for your board version from that page and flash it.
[image]

Nbrooks639, 2022-03-07:
No s1  Sent from Mail for Windows From: Miguel Risco-CastilloSent: Monday, March 7, 2022 12:49 AMTo: mriscoc/Marlin_ConfigurationsCc: Nbrooks639; ManualSubject: Re: [mriscoc/Marlin_Configurations] Support for "AutoLev" sensor from Substancia 3D (Issue #1) Download the appropriate binary file for your board version from that page and flash it.

closed 2022-06-08 by mriscoc