mriscoc/Special_Configurations #2: Combination with Engraver "Firmware"?

Cyrelion, 2022-02-15:
First of all, I absolutely love your firmware binaries, and they are doing a really excellent job, but I had to switch to the original firmware after I got the Creality CV-LaserModul for my Ender 3 S1 :(

Do you think it would be possible to combine both, or enable some engraving features (the M03 command) in your version as well?

This would be a huge benefit for me so I can stay on your firmware.

Thanks in advanced,
Cyrelion

mriscoc, 2022-03-04:
The official firmware sources for the Ender3S1 have been published, the last version of the Professional Firmware uses the configuration data from those sources, then probably the Creality CV-LaserModul will now work.

Cyrelion, 2022-03-04:
Thank you for the update. I've installed it just a second ago and I can't switch mode between 3d and Laser mode, so it looks like it's not working :(

Can you send me the link to the official firmware sources so I can ask there for the values to set?

mriscoc, 2022-03-04:
Here are the official sources: https://github.com/CrealityOfficial/Ender-3S1

Cyrelion, 2022-03-06:
In the "official" CV-LaserModul Firmware (which is a different one than the Ender3-S1 bin file you have a menu in the Control
[image]
[image]
[image]
[image]

mriscoc, 2022-03-06:
That is another firmware not the released one. But if the Laser module uses a fan port maybe it can be controlled using the commands:
```gcode
M106 ; Laser on command
M107 ; Laser off command
```

Cyrelion, 2022-03-07:
I know, that's the problem :) But it will not work with your actual release out of the box, because if you switch the printing head to the laser head it will remove the z-endstop/CL-Touch sensor too, so you don't have a z-axis stop anymore.

Cyrelion, 2022-03-07:
Yeah, Creality answered my request for the sources and uploaded the modification in a branch of the Ender 3 S1:

https://github.com/CrealityOfficial/Ender-3S1/tree/ender-3s1-lasermodel

mriscoc, 2022-03-07:
Great!

Cyrelion, 2022-03-07:
I hope you can implement that into your version of the Marlin firmware :)

mriscoc, 2022-03-07:
I see the sources and there are a lot of changes, that is why they provide another firmware to use the laser engraver. I don't know if I will implement all the changes because I don't want to increase the size if the firmware too much for functions that are not to all.

Cyrelion, 2022-03-07:
Yeah, understand that but still sad, so I have to continue to flash the firmware after changing the printhead. But this is working too not as easy as this but possible

mriscoc, 2022-03-08:
> I know, that's the problem :) But it will not work with your actual release out of the box, because if you switch the printing head to the laser head it will remove the z-endstop/CL-Touch sensor too, so you don't have a z-axis stop anymore.

I have a S1 manual mesh version that uses the Z end-stop port: https://github.com/mriscoc/Marlin_Configurations/releases/tag/S1-ManualMesh

Cyrelion, 2022-03-08:
> I have a S1 manual mesh version that uses the Z end-stop port: https://github.com/mriscoc/Marlin_Configurations/releases/tag/S1-ManualMesh

That will not work either because you don’t have an endstop at all 😀

mriscoc, 2022-03-08:
You can connect the Z limit switch to use the S1 Manual mesh.

mriscoc, 2022-04-06:
Could be possible to use the G92 to set the Z = 0 position without do a Z home.

Cyrelion, 2022-04-06:
The main problem would be to disable the Z-Homing at all (which is something the laser firmware is doing.)

There are a complete different menu for engraving including z-axis movement which is done without leveling etc. Maybe I can try to make a video of the menus with laser firmware and going through all the "features" that is visible on that firmware version if that helps