mriscoc/Special_Configurations #4: Support for 100k 3950 Sensor with BL/CRTouch and stock hotend

eblieb, 2022-02-28:
I have the Ender 3 V2 4.2.2 board with a M3 Hex 100k 3950 Temperature prob that I need the sensor changed to 13.
https://www.amazon.com/gp/product/B08YNX79H8/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1  - This is the sensor I am using. 

What else do I need to get it built? 

{
"Configuration_adv.h" : [
],
"Configuration.h" : [
  {
    "op": "CustomVal",
    "searchfor": "TEMP_SENSOR_0",
    "value": "13",
    "comment": "100k Hisens 3950 1% up to 300°C"
  },
],
"Version.h" : [
]   
}

mriscoc, 2022-03-01:
Configuration files for 3950T
[422-BLTouch-3950T.zip](https://github.com/mriscoc/Marlin_Configurations/files/8158425/422-BLTouch-3950T.zip)
You can use these configuration files to compile your own firmware.
Precompiled Firmware:
[Ender3v2-3950T-20220228.zip](https://github.com/mriscoc/Marlin_Configurations/files/8158460/Ender3v2-3950T-20220228.zip)

eblieb, 2022-03-01:
Thank you so much. Guess I have to build a build environment soon

eblieb, 2022-03-02:
In the config file 
#define LIMITED_MAX_FR_EDITING    
and
  #define LIMITED_JERK_EDITING

were both commented out which caused the latest source to fail compile. Is there a reason these are commented out?

eblieb, 2022-03-02:
Also MAX_ACCEL_EDIT_VALUES

mriscoc, 2022-03-03:
Yes, that is a recent change from the last release, the firmware is now using those configuration values. You can download the new configuration files from the repository: https://github.com/mriscoc/Marlin_Configurations/tree/main/422-BLTouch-3950T

eblieb, 2022-03-03:
I just uncommented them from the configuration i had, is that all that changed? Dont want to rebuild if i dont have to right now

mriscoc, 2022-03-03:
If you are customizing the configuration files it is better to move those custom values to a Json and apply using the provided scripts then you will always have up to date version.

eblieb, 2022-03-04:
Is there any write ups on how exactly to apply the JSON to the source code?

mriscoc, 2022-03-04:
The Json is used by the Python script to mod the original configuration files from the bugfix branch of Marlin, that allows you to always have up to date configuration files if the original ones are changed. Take a look at 3950T.json and V2-3950T.py
```python
import CreateConfigs
CreateConfigs.Generate('', ['422','BLTouch','3950T'])
```
These lines create for you the configuration files for Ender3v2 with a 4.2.2 control board, BLTouch and a 3950 sensor.

mriscoc, 2022-03-14:
AM means All-Metal, if you don't have an All-Metal hot end use the configuration without the AM.
https://github.com/mriscoc/Marlin_Configurations/tree/main/Ender3V2-422BLT-3950

closed 2022-06-08 by mriscoc