mriscoc/Special_Configurations #6: Support for Adafruit Neopixels - Digital RGB LEDs

mriscoc, 2022-03-07:
Preliminary support for Adafruit Neopixels - Digital RGB LEDs

**Important** Set the NEOPIXEL_PIN to suit your mainboard.

https://teachingtechyt.github.io/upgrades.html#neopixels

```json
{
"Configuration_adv.h" : [
],
"Configuration.h" : [
  {
    "op": "Enable",
    "searchfor": "NEOPIXEL_LED",
    "comment": "Adafruit Neopixels"
  },
  {
    "op": "Custom",
    "searchfor": "NEOPIXEL_TYPE",
    "mask" : ".*",
    "value": "NEO_GRB",
    "comment": "Adafruit Neopixels"
  },
  {
    "op": "Custom",
    "searchfor": "NEOPIXEL_PIN",
    "mask" : ".*",
    "value": "PB2",
    "comment": "Adafruit Neopixels"
  },
  {
    "op": "CustomVal",
    "searchfor": "NEOPIXEL_PIXELS",
    "value": "15",
    "comment": "Adafruit Neopixels"
  },
  {
    "op": "Enable",
    "searchfor": "NEOPIXEL_STARTUP_TEST",
    "comment": "Adafruit Neopixels"
  },
  {
    "op": "Enable",
    "searchfor": "NEOPIXEL_EVENT_LEDS",
    "comment": "Adafruit Neopixels"
  }
],
"Version.h" : [
]   
}
```

https://youtu.be/SNAepJUDlsA

Nbrooks639, 2022-04-01:
if adding neopixels to the ENDER 3 S1 you only have two pins that are open and not sharing with something else.

[image]

you also cannot use the expansion connection behind z stepper motor. power needs to come from power supply and be stepped down to 5v via buck convertor or step down regulator. 
the led control menu will not work in this firmware so you need to enable EVENT_PRINTER_LIGHTS

mriscoc, 2022-04-02:
> if adding neopixels to the ENDER 3 S1 you only have two pins that are open and not sharing with something else.
> 
The Neopixel uses one pin for data, are you using the PA13 pin?  For what is PA14?

Nbrooks639, 2022-04-02:
you have the option to use either PA13 or PA14 for the data pin. sorry i should have specified that. either one with work as long as you change in configs.

mriscoc, 2022-04-02:
Thanks for the info. I'm going to merge the light control menu already added to the ProUI configuration in the official Marlin repository, but it will not be enabled by default, if you compile the firmware you must enabled to use it.

Nbrooks639, 2022-04-02:
Cool. I'm glad you decide this after I have already done it all lol. Sent from my T-Mobile 5G Device

closed 2022-06-08 by mriscoc