mriscoc/Special_Configurations #7: Ender 3 V2 450degree thermistor hi temp hot end

JCRodzDom, 2022-03-08:
Hi Miguel, Ive been trying to compile firmware for my 3D printer long time and I cant success. 
Needing help on building the firmware for a Ender 3 V2, 4.2.7 board, slice enginering 450 degree thermistor.
I used your firmware Ender 3V2 /S1 and works ok, but when I changed the thermistor ( 67 : 500kΩ  SliceEngineering 450°C Thermistor )  
( #define TEMP_SENSOR_0 67 ), the actual temperature reading is not adequate. When is room temperature reads 275 degrees. So I install the original thermistor and reads room temperature correctly.  How I can correct and get a real reading from the new thermistor ?
Thanks in advance for your help.

JCRodzDom, 2022-03-08:
Just to let you know im trying to print HTN CF25 filament. This is to print and airfoil of a wing. In other words im printing RC glider wings.

mriscoc, 2022-03-08:
For use that thermistor, you can use the Configuration files from the Volcano version, change the number in TEMP_SENSOR_0 and increase the temperature in HEATER_0_MAXTEMP. Then overwrite the configuration files inside of Marlin folder from the main project files are in the firmware repository: https://github.com/mriscoc/Ender3V2S1

JCRodzDom, 2022-03-09:
thanks a lot for your info, so I test the thermistor from slice eng. at 25 degrees Celsius and the resistance was 1091 ohms meaning that the table provided by Slice and the table at Marlin is not compatible with what Slice eng is selling right now. It looks that the 450 degree thermistor is out of stock and they replace it by a RTD Pt1000.

JCRodzDom, 2022-03-09:
Hi Miguel, sorry to keep bothering. So i found the thermistor temp table but I need know where i can find the table for thermistor 67 in order to put my values and have a real reading. I think they are in THERMISTORNAMES.h ; but i cant find it anywhere.

[image]

JCRodzDom, 2022-03-10:
Hola Miguel
So i figure it out with the thermistor, so for the Slice Engineering RDT Pt1000 the thermistor in Marlin is the number 1047 that is already programed.
Now, i compiled your firmware but is giving me a WARNING like the home is not found and the servos are not moving at all.
Do you think that this is related to the 427 board?
What im missing?
let me know thanks

mriscoc, 2022-03-10:
Check if you are using the correct configuration for your board:

```cpp
  #define MOTHERBOARD BOARD_CREALITY_V427  // Board v4.2.7
```

mriscoc, 2022-04-14:
No more feedback, closing the issue.

closed 2022-04-14 by mriscoc