mriscoc/Special_Configurations #8: Latest version here: https://github.com/mriscoc/Marlin_Configurations/releases/tag/AutoLev

RPI-PMK, 2022-04-13:
Latest version here: https://github.com/mriscoc/Marlin_Configurations/releases/tag/AutoLev
At the end of the page, Assets section.

_Originally posted by @mriscoc in https://github.com/mriscoc/Special_Configurations/issues/1#issuecomment-1060201223_

Sorry, could not work out how to post feedback on original post so opened this one.

New firmware installed and working well.

Offsets needed when using standard bracket are:

X = -42.0mm
Y = 10.0mm
z = -0.2mm

Auto mesh works and prints so far looking good too.

RPI-PMK, 2022-04-22:
Hi, has this version been updated with latest April release. I am just about to start a few long prints so can test fully.

Thanks

mriscoc, 2022-04-25:
Sorry, not all the special versions are updated at the same time. But it is easy to compile them by yourself, you need only the already provided configuration files and the source files from the main repository.

closed 2022-06-08 by mriscoc