mriscoc/Special_Configurations #13: Issue compiling for skr mini e3v3

nate-zero, 2022-07-25:
I'm having trouble getting this latest version to compile for an skr mini e3v3

The error appears to be regarding filament runout, can you please advise?

[Marlin.zip](https://github.com/mriscoc/Special_Configurations/files/9184894/Marlin.zip)

mriscoc, 2022-07-25:
We need more information, what error did you get in Visual Studio Code.

nate-zero, 2022-07-25:
Compiling
.pio\build\STM32G0B1RE_btt\src\src\gcode\feature\trinamic\M122.cpp.o
In file included from
Marlin\src\gcode\feature\runout\../../../feature/../module/stepper/../../feature/../lcd/e3v2/proui/dwin_lcd.h:24,
                 from
Marlin\src\gcode\feature\runout\../../../feature/../module/stepper/../../feature/../lcd/e3v2/proui/dwinui.h:28,
                 from
Marlin\src\gcode\feature\runout\../../../feature/../module/stepper/../../feature/../lcd/e3v2/proui/dwin.h:26,
                 from
Marlin\src\gcode\feature\runout\../../../feature/../module/stepper/../../feature/../lcd/marlinui.h:50,
                 from
Marlin\src\gcode\feature\runout\../../../feature/../module/stepper/../../feature/tmc_util.h:25,
                 from
Marlin\src\gcode\feature\runout\../../../feature/../module/stepper/trinamic.h:35,
                 from
Marlin\src\gcode\feature\runout\../../../feature/../module/stepper/indirection.h:44,
                 from
Marlin\src\gcode\feature\runout\../../../feature/../module/stepper.h:47,
                 from
Marlin\src\gcode\feature\runout\../../../feature/runout.h:31,
                 from Marlin\src\gcode\feature\runout\M412.cpp:28:
Marlin\src\gcode\feature\runout\../../../feature/../module/stepper/../../feature/../lcd/e3v2/proui/../common/dwin_api.h:24:10:
fatal error: ../../../inc/MarlinConfig.h: No such file or directory
   24 | #include "../../../inc/MarlinConfig.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\STM32G0B1RE_btt\src\src\gcode\feature\runout\M412.cpp.o]
Error 1
=========================================================== [FAILED] Took
80.39 seconds ===========================================================

Environment      Status    Duration
---------------  --------  ------------
STM32G0B1RE_btt  FAILED    00:01:20.387
====================================================== 1 failed, 0
succeeded in 00:01:20.387
======================================================

mriscoc, 2022-07-25:
Use the source files from here: https://github.com/mriscoc/Ender3V2S1 with your configuration files the firmware compiles ok.

mriscoc, 2022-08-02:
No more feedback

closed 2022-08-02 by mriscoc