mriscoc/Special_Configurations #14: No 300 degrees

MathyCry, 2022-08-02:
Hello, I successfully flashed this firmware onto my board. But can only set 260 degrees.

xx1180, 2022-08-02:
https://www.youtube.com/watch?v=LyU2DCCP5Is&t=95s
watch this video it explains what you need to do it worked for me but please remember that after running this you need to store the settings or when you turn off it will revert back to max 260c

mriscoc, 2022-08-02:
CAUTION!. Be sure that your hot-end really supports 300°C and there isn't any PTFE in any place (as terminal isolators for the thermistor for example)

mriscoc, 2022-08-02:
Because this is not a firmware error, I will close this issue.

closed 2022-08-02 by mriscoc

MathyCry, 2022-08-02:
I followed the instructions from the video. But it doesn't work. I can still only set it to 260°C

xx1180, 2022-08-02:
did you use notepad++ or just notepad

MathyCry, 2022-08-02:
OS X TextEdit I used

MathyCry, 2022-08-02:
Sorry, I found the error. Thank you very much for your help, you are great.