mriscoc/Special_Configurations #16: Preliminary experimental support for Voxelab Aquila printers

mriscoc, 2022-08-21:
Caution, this is a very experimental support.

This firmware needs these display assets: https://github.com/mriscoc/Ender3V2S1/wiki/How-to-update-the-display

[image]

## Donations
Thank you for your support, I receive donations through [Patreon](https://www.patreon.com/mriscoc) and [Paypal](https://www.paypal.com/paypalme/mriscoc)   

[[image]](https://www.paypal.com/donate?business=85SPAAR6UZEE8&currency_code=USD)

## Disclaimer
THIS FIRMWARE AND ALL OTHER FILES IN THE DOWNLOAD ARE PROVIDED FREE OF CHARGE WITH NO WARRANTY OR GUARANTEE. SUPPORT IS NOT INCLUDED JUST BECAUSE YOU DOWNLOADED THE FIRMWARE. WE ARE NOT LIABLE FOR ANY DAMAGE TO YOUR PRINTER, PERSON, OR ANY OTHER PROPERTY DUE TO USE OF THIS FIRMWARE. IF YOU DO NOT AGREE TO THESE TERMS THEN DO NOT USE THE FIRMWARE.

labotecno, 2022-08-31:
how i can to install it ? with windows?

mriscoc, 2022-08-31:
> how i can to install it ? with windows?

The configurator only needs Python 3 installed.

doctorevil30564, 2022-11-22:
any chance for a N32 variant?

mriscoc, 2022-11-23:
> any chance for a N32 variant?

Only if we get many positive tested configuration files.