mriscoc/Special_Configurations #18: Ender3V2-427-BLTUBL-LA-20221002.bin not loading

Zyzzbarth333, 2022-10-27:
Hi there, Ender3V2-427-BLTUBL-LA-20221002.bin, seems to give a blank screen when trying to laod it onto my printer, I reloaded current version of ENDER3V2-427-BLTUBL and works fine.

SargFowler, 2022-10-30:
I can confirm the same issue. The bed gets hot and the temp runs away - mine got up to 100c.
The non LA version works fine.

mriscoc, 2022-11-01:
Rename the firmware to a shorter name, change the filename between attempts.

SargFowler, 2022-11-03:
Thanks for the reply. I've tried what you've suggested, but the BL just clicks away and there's no display. Other version works fine. How can I help to diagnose the fault?

mriscoc, 2022-11-04:
Please test this: https://github.com/mriscoc/Special_Configurations/releases/download/LinearAdvance/Ender3V2-427-BLTUBL-LA-20221019.bin

SargFowler, 2022-11-04:
Fabulous, that loads and runs OK. Just need to tune in the values to get the best print.
Thanks Miguel

Zyzzbarth333, 2022-11-05:
Thank you Miguel! Works perfectly now!

mriscoc, 2022-11-05:
Thank you for the confirmation.

closed 2022-11-05 by mriscoc