mriscoc/Special_Configurations #19: No motor movement with "Ender 3 v2/4.2.7 MB/Sprite Pro/CR Touch" hardware config

redoxeon, 2022-11-04:
**Description**
I installed [Experimental-Ender3V2-427-BLTUBL-T13-20221002](https://github.com/mriscoc/Special_Configurations/releases/tag/T13) and everything in the guides seems to go according to the documentation, but when I try to manually move any motor or auto-home/park it doesn't move and I eventually get the message "Printer killed! Turn off printer."

**To Reproduce**
Steps to reproduce the behavior:
1. Install Experimental-Ender3V2-427-BLTUBL-T13-20221002 on an Ender 3 v2/4.2.7 MB/Sprite Pro/CR Touch hardware config
2. Reset settings to default then store, reboot (not sure if this is important or not, but it was in the guide, and I did it)
3. Set Esteps, store, try to auto-home
4. wait until you're notified that your printer has been killed

**Expected Behavior**
Auto-home works

**Version**
[Experimental-Ender3V2-427-BLTUBL-T13-20221002](https://github.com/mriscoc/Special_Configurations/releases/tag/T13) Released Oct 2, 2022

**Additional Context**
I also tried and failed to reinstall, even after renaming the file a few times. Maybe I did it wrong? It just boots normally.

mgadei, 2022-11-09:
could we have a T13 but LA version ?

mriscoc, 2022-11-10:
Sorry,  seems to be an error in my compile chain. The 427 version was compiled by mistake with the 4.2.2 drivers pin out.

mriscoc, 2022-11-10:
Please try this version:
[Experimental-Ender3V2-427-BLTUBL-T13-20221026.zip](https://github.com/mriscoc/Special_Configurations/files/9977394/Experimental-Ender3V2-427-BLTUBL-T13-20221026.zip)

mriscoc, 2022-11-10:
> could we have a T13 but LA version ?

The precompiled Special Versions are sponsored versions, but you can compile your own using the configurator Python script. https://github.com/mriscoc/Special_Configurations

redoxeon, 2022-11-10:
In the middle of a 24 hour print (currently back on stock firmware to try to catch up on my personal backlog) but I'll flash it once it finishes 😄

mriscoc, 2022-12-25:
No more user feedback

closed 2022-12-25 by mriscoc