mriscoc/Special_Configurations #21: Ender 3 S1 la & ubl

cooltrooper, 2022-12-04:
Loving your firmware!
I have an s1 and the ubl in the standard build is very good. Came to look for the linear advance version and noticed it doesn't have ubl?

A UBL and LA build for ender v3 s1 would be ideal.

**Describe alternatives you've considered**
I haven't braved building it from source myself.

mriscoc, 2022-12-05:
Ender 3 S1 with F1 SoC has UBL and LA: https://github.com/mriscoc/Special_Configurations/releases/download/LinearAdvance/Ender3S1-F1-UBL-LA-20221002.bin
Ender 3 S1 with F4 SoC does not have enough program memory to enable many features.

closed 2022-12-05 by mriscoc

cooltrooper, 2022-12-05:
Thank you! I have the f4 :(

mriscoc, 2022-12-05:
> Thank you! I have the f4 :( Does the la version have the stock style bed leveling enabled?

Yes, ABL is enabled.