mriscoc/Special_Configurations #23: Marlin Build fails with SKR Mini Board and no special features

fabiomazzone, 2022-12-25:
**Describe the bug**
There is a build error if you try compile a firmware version for the ender 3 v2 with the SKR mini v3 board and no special features.

```
Compiling .pio/build/STM32G0B1RE_btt/FrameworkArduino/stm32/usb/usbd_ep_conf.c.o
Compiling .pio/build/STM32G0B1RE_btt/FrameworkArduino/stm32/usb/usbd_if.c.o
Compiling .pio/build/STM32G0B1RE_btt/FrameworkArduino/wiring_analog.c.o
Compiling .pio/build/STM32G0B1RE_btt/FrameworkArduino/wiring_digital.c.o
Compiling .pio/build/STM32G0B1RE_btt/FrameworkArduino/wiring_pulse.cpp.o
Compiling .pio/build/STM32G0B1RE_btt/FrameworkArduino/wiring_shift.c.o
Compiling .pio/build/STM32G0B1RE_btt/FrameworkArduino/wiring_time.c.o
Archiving .pio/build/STM32G0B1RE_btt/libFrameworkArduino.a
Indexing .pio/build/STM32G0B1RE_btt/libFrameworkArduino.a
Linking .pio/build/STM32G0B1RE_btt/firmware-20221225-181111.elf
/home/fabiomazzone/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: Marlin/lib/proui/libproui.a(proui.cpp.o): in function `ManualMesh_Resize()':
proui.cpp:(.text._Z17ManualMesh_Resizev+0xb8): undefined reference to `mesh_bed_leveling::index_to_ypos'
/home/fabiomazzone/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: proui.cpp:(.text._Z17ManualMesh_Resizev+0xbc): undefined reference to `mesh_bed_leveling::index_to_xpos'
collect2: error: ld returned 1 exit status
scons: *** [.pio/build/STM32G0B1RE_btt/firmware-20221225-181111.elf] Error 1
===================================================================================================================== [FAILED] Took 73.35 seconds =====================================================================================================================

Environment      Status    Duration
---------------  --------  ------------
STM32G0B1RE_btt  FAILED    00:01:13.352
================================================================================================================ 1 failed, 0 succeeded in 00:01:13.352 ================================================================================================================
```

**To Reproduce**
Steps to reproduce the behavior:
1. Create  a configuration with ```CreateConfigs.Generate('Ender3V2-SKRME3V3', ['Ender3V2', 'SKRME3V3'])```
2. copy the configuration files to the marlin folder in the ender3v2a1 repo
3. run ```platformio run -e STM32G0B1RE_btt```
4. See error

**Expected behavior**
Having an successfull build

mriscoc, 2022-12-25:
You must select a leveling system and thermistor type:
```python
CreateConfigs.Generate('Ender3V2-SKRME3V3-MM',['Ender3V2','SKRME3V3','MM'])
```

fabiomazzone, 2022-12-25:
Ahhh thats what MM stands for, I got the default Marlin Firmware running for now, to print the parts for a clicky probe. 
Probably after that I am coming back to your setup.

closed 2022-12-25 by fabiomazzone