mriscoc/Special_Configurations #31: Support for F4 chip with UBL and 300c changes

EnderSyth, 2023-01-21:
Currently there is no binaries that have support for both the UBL and 300c support for the Sprite Pro Extruder for the F4 chip. I specifically have an Ender 3 S1 Plus and have been trying to find a way to get the full support, and while there is an F1 chip UBL with 300c support its missing for the F4 variant. There is, in the main release, a F4 UBL variant of the binaries with the standard max temp of 260c.

I'm hoping since in this special config it has an F1 build with the UBL and the 300c support that such support can be extended to the F4 chip variant.

mriscoc, 2023-01-22:
https://github.com/mriscoc/Special_Configurations

EnderSyth, 2023-01-22:
I've not yet built a custom firmware version myself and was hoping it would be a setting that could be set on the machine via gcode or inside the advance settings (despite finding nothing of the sorts in my searches). I'm guessing since you've linked to there that it means definitively that such change requires a customized firmware to be built. 

Given how UBL seems to becoming more common I might just wait for the next release and hope there is a build compatible, but not sure I want to dive into building a custom variant of the firmware just yet myself. Thank you for the answer.

mriscoc, 2023-01-22:
Stock Ender 3S1 extruder does not have support for 300°C only the Sprite Extruder **PRO** has support for 300°C, in that case the needed firmware is the T13 version.

EnderSyth, 2023-01-22:
I do have the pro extruder, I bought it and got the v2 touchscreen and this firmware so i could make use of it. Creality offered me the s1 pro firmware but said I’d be limited to its 200x200 bed size if I used it. 

I didn’t realize that the F4 chip had less memory. That is a pain and kinda annoying since the s1 plus is quite pricey, why must they have cheaped out on the mainboard. Oh well thank you for the information.

mriscoc, 2023-01-22:
> I didn’t realize that the F4 chip had less memory. That is a pain and kinda annoying since the s1 plus is quite pricey, why must they have cheaped out on the mainboard. Oh well thank you for the information.

Yes, that's why is better to compile or request a very customized version of the F4 because each user has to decide which features wants to disable, for example, the F4 UBL version has these features disabled: https://github.com/mriscoc/Ender3V2S1/issues/648

EnderSyth, 2023-01-22:
Wow that is quite a long list of features to disable to fit UBL. Seems like it might be worth looking at a replacement motherboard in the future if I expect to get any more advanced features. Shame you can’t just expand the memory of the chip somehow.

Raxer10, 2023-01-26:
Hello, I am also the owner of an Ender S1 Pro. 
The T-13 version works perfectly! Depending on how straight the print bed is, it is enough to just level with the Tramming Wizard!
 For owners of a newer printer, within the warranty, I recommend that if the print bed is not straight, you complain to Creality about this print bed and make it absolutely clear that you will otherwise send the printer back!! 
Surprisingly, the spare part delivered is thicker than the original and above all straight :-) 
When unlocking the temperature with the custom G-code, the temperature is set to 315 degrees, then you can heat it to 300 degrees like the original ^^

mriscoc it is possible to install a custom G-code in the firmware that increases the temperature of the -print bed- like the temperature of the hotend in the way:
 C104 U1 T###: unlocks the hotend maximum temperature to ###°C (CAUTION) ????
Because the Ender S1 Pro can heat the heated bed up to 110 degrees

mriscoc, 2023-01-27:
> Because the Ender S1 Pro can heat the heated bed up to 110 degrees

I think that latest version allows you to set the bed temperature to 110 °C

Raxer10, 2023-01-28:
Sorry for my mistake was my fault ;-) Yes it is in the new Version i habe not tested it with it.. Thank you for the update :-)

closed 2023-01-30 by mriscoc