mriscoc/Special_Configurations #35: Missing k-factor setting in print tune menu

smensak, 2023-03-18:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

Updated to Experimental-Ender3S1-F4-LA-20230312.bin and the k-factor setting in print tune menu is no longer present. Was there in previous 2022 release. 

### Bug Timeline

2023/03/12

### Expected behavior

K-factor value setting present

### Actual behavior

K-factor value setting not present

### Steps to Reproduce

Start a print, go to tune and you wont find k-factor setting.

### Version of Professional Firmware

2023/03/12 F4 LA

### Printer model

Ender 3 S1

### Electronics

F4

### Add-ons

none

### Bed Leveling

None

### Your Slicer

Prusa Slicer

### Host Software

None

mriscoc, 2023-03-18:
Due to space constraints in the Tune menu, the K-factor is now only present in the Control/Motion menu, but it can be re-enabled in a custom build if needed.

closed 2023-03-18 by mriscoc

smensak, 2023-03-18:
Thanks for the info, where do I look for it?

mriscoc, 2023-03-18:
Enable this
https://github.com/mriscoc/Special_Configurations/blob/a213d830f80cc75ebdd4bddc6b4429f4449180ae/Ender3S1-F4-LA/Configuration.h#L3288